clsergent/Snapmaker2Postprocessor #1: Error with version 0.20.2

spitzlbergerj, 2023-01-04:
Hi 
Thanks a lot for your work. I just found your repositiory and tried it with the actual Version 0.20.2 27.12.2022

here are my versions

`
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * LCInterlocking
`

I got this error messages:
`
16:21:19  post: Snapmaker2(D:/OneDrives/OneDrive - La Gondola Barocca/_PCBüroLustheim/Eigene Dateien/CAD, Designer, Publisher/_CNC-Fräsungen/Vakuumplatte/vakuumplatte200x200.clsergentSnapmaker.cnc, )
16:21:19  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 640, in export
    post.export(objects, filename, argstring)
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 546, in export
    if self.conf.thumbnail and (thumbnail := getThumbnail(self.job)):
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 109, in getThumbnail
    for model in job.Model.Group:

'NoneType' object has no attribute 'Model'
`

Thanks for having a lot at this.

Josef

clsergent, 2023-01-04:
Hi,

Thanks for opening this issue.
I have not been able to reproduce the bug so far, but it seems that the script is unable to retrieve any Path job (PathJob.ObjectJob).

Here are several questions to help me find the bug :
1) In the report view, do you get any error such as "No parent job was found for {obj}, returning selected job"? If so could you post the obj name provided?
2) have you selected any job when clicking on the post processor button? If not, could you please retry with a job selected?

clsergent, 2023-01-04:
In the meantime, you can add `--no-thumbnail` in the Post Processor Args field of your job, which will skip the thumbnail generation and should avoid the error you encountered.

clsergent, 2023-01-04:
I finally reproduced your error. It happens when you select an operation rather than a job before calling the post processor.
If you select a job, it should work seamless. However, you may want to generate gcode for a single operation, so I will publish a patch.

spitzlbergerj, 2023-01-05:
Hi

thank you very much for the quick analysis and the patch. I must admit that I am a newcomer to CNC and this is the first time I have used the Path workbench and the Post Processors. Therefore the problem is probably in front of the computer and not in the code.

I have installed the latest version of the script and tried several things:

1. --no-thumbnail argument
the gcode is generated!
but I get a whole series of messages like:
```
08:19:28 Command ignored: M6
08:19:28 G0: negative speed inserted (0.0)
```

2. no argument, job selected (as before)
no gcode generated
last message: 
```
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 117, in getThumbnail
    FreeCADGui.Selection.addSelection(obj.Document.Name, obj.Name)

Gui.SelectionObject' object has no attribute 'Name'.
```

3. operation selected
no gcode generated
last message: 
```
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 117, in getThumbnail.
    FreeCADGui.Selection.addSelection(obj.Document.Name, obj.Name)

Gui.SelectionObject' object has no attribute 'Name'.
```

4. a single operation selected
no gcode generated
last message: 
```
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 117, in getThumbnail
    FreeCADGui.Selection.addSelection(obj.Document.Name, obj.Name)

Gui.SelectionObject' object has no attribute 'Name'.
```

If I insert --no-thumbnail in 3. and 4. I get a gcode in each case (with the error messages about the speed as above). 

in the report view no error message is displayed as you asked. 

My FreeCAD model is [here ](https://lagondolabarocca-my.sharepoint.com/:u:/g/personal/josef_spitzlberger_de/EfdQa4ahT3dCo9fqJBg-AWcBD31X7DsmWGyAVjyTk6gMmQ?e=QDg7qR)

I hope this information helps

Many thanks again

Sepp

clsergent, 2023-01-05:
Hi,

Thanks for your report, we are moving forward!

1.  `--no-thumbnail` result:
- The gcode produced confirms that the error is related to the thumbnail generation. It seems to work otherwise.

- The various messages you get are warnings only.  I will update the logging system to use `FreeCAD.Console` rather than standard library module `logging`. Warnings will appear in orange rather than red.

  - `Command ignored: M6`: Snapmaker gcode does not include M6 (tool change). The postprocessor adds an HMI pause (M76) but also warns about the removal of M6 commands.

  - `G0: negative speed inserted (0.0)`: the postprocessor warns about negative (or null, I will correct the warning...) rapid speed. It means that you have not specified any rapid horizontal/vertical speed. You should configure them in the Tools tab :
      [image]

2. job or operation selection:
- I have not been able to reproduce your error using your file. However, your logs indicate that the script is not able to properly restore the selection. Indeed, to generate the thumbnail, it manipulates the selection to remove unnecessary artefacts from the active view while recording the thumbnail, then restores the selection.
- Could you try [this custom version](https://github.com/clsergent/Snapmaker2Postprocessor/files/10354312/Snapmaker2_post_custom.py.zip) of the postprocessor and report the logs?

spitzlbergerj, 2023-01-06:
Hi and good morning from Bavaria,

Thanks a lot for your help and efforts!

I did the tests with the latest version from the repository (latest commit 9e3fd68). If you want me to test with the intermediate version too, please let me know.

Changes to my model or to FreeCAD:
- Importing the new version of the Post Processor
- Job - Output File: only a directory specified, no longer an explicit filename
- Horiz. / Vert. Rapid Speed set to 500 mm/min via the Setup Sheet

**Test result WITH parameter --no-thumbnail:**
Selection of Job, Operations or a specific operation:
_gcode created in each case, one warning in each case_

```
10:16:48 post: Snapmaker2(D:/OneDrives/OneDrive - La Gondola Barocca/_PCBüroLustheim/Own Files/CAD, Designer, Publisher/_CNC-Millings/Vacuum-Plate/xyz.cnc, --no-thumbnail)
10:16:48 Post Processor: Snapmaker2_post
Postprocessing...
10:16:48 Command ignored: M6
10:16:48 Postprocessing done
```

Attention: please see note on error occurring when double-clicking on "Job" below !!

**Test result WITHOUT parameter --no-thumbnail**
Select job, operations or a specific operation:
_gcode NOT created, one error message

```
10:21:14 Post: Snapmaker2(D:/OneDrives/OneDrive - La Gondola Barocca/_PCBüroLustheim/Own files/CAD, Designer, Publisher/_CNC-Millings/xyz.cnc, )
10:21:14 Post Processor: Snapmaker2_post
Postprocessing...
10:21:14 C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py(119)<class 'RuntimeError'>: Unknown file exception happened
```

My workflow:
- Starting FreeCAD (see below for version)
- open the file
- change the workbench to Path
- click on Job (Job is selected) or the other elements
- click on Post Process
- Appearance of the file dialog with input of a file name
- Appearance of the messages above (warning or error)

my FreeCAD version: (still the same)
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * LCInterlocking
```

While testing around, I noticed that when I double click on Job, a series of error messages appear. I can't tell if this was the case yesterday as well. I mean "rather not", but I am not sure anymore.

The error messages always appear (directly after starting Freecad and loading the file and the Workbench or also afterwards sometime), if I double-click on the Job to call the job dialog.

```
10:26:05 Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftutils\todo.py", line 139, in doTasks
    f(arg)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftmake\make_clone.py", line 129, in <lambda>
    ToDo.delay(lambda col: setattr(cl.ViewObject, "DiffuseColor", col),
ReferenceError: Cannot access attribute 'ViewObject' of deleted object

10:26:05 ToDo.doTasks, Unexpected error:
<class 'ReferenceError'>
in <function make_clone.<locals>.<lambda> at 0x0000025DBF33A4D0>([(0.800000011920929, 0.800000011920929, 0.800000011920929, 0.0)])
10:26:05 Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftutils\todo.py", line 139, in doTasks
    f(arg)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftmake\make_clone.py", line 129, in <lambda>
    ToDo.delay(lambda col: setattr(cl.ViewObject, "DiffuseColor", col),
ReferenceError: Cannot access attribute 'ViewObject' of deleted object

10:26:05 ToDo.doTasks, Unexpected error:
<class 'ReferenceError'>
in <function make_clone.<locals>.<lambda> at 0x0000025DBF33A560>([(0.800000011920929, 0.800000011920929, 0.800000011920929, 0.0)])
10:26:05 Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftutils\todo.py", line 139, in doTasks
    f(arg)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Draft\draftmake\make_clone.py", line 129, in <lambda>
    ToDo.delay(lambda col: setattr(cl.ViewObject, "DiffuseColor", col),
ReferenceError: Cannot access attribute 'ViewObject' of deleted object

10:26:05 ToDo.doTasks, Unexpected error:
<class 'ReferenceError'>
in <function make_clone.<locals>.<lambda> at 0x0000025DCFCABD90>([(0.800000011920929, 0.800000011920929, 0.800000011920929, 0.0)])
```

I will now create a completely new, very simple model and test everything again. Possibly my model contains a bug that is causing all this.

Once again, thank you very much for your help. If there is an ''Amazon Wish List'' or a ''Coffee Donation'' I can make, please let me know. I know from own experience how much time maintaining a Github repository can eat up.

Many greetings

Sepp

spitzlbergerj, 2023-01-06:
Hi,
I just tried a very simple, completely new model (cuboid with a pocket with rounded corners). 

With the --no-thumbnail parameter the gcode is created without error message. Without the parameter I get again the error from above

```
11:12:06 post: Snapmaker2(D:/OneDrives/OneDrive - La Gondola Barocca/_PCBüroLustheim/Own files/CAD, Designer, Publisher/_CNC-Millings/TestModel CNC/testModel-thumbnail.cnc, )
11:12:06 Post Processor: Snapmaker2_post
Postprocessing...
11:12:06 C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py(119)<class 'RuntimeError'>: Unknown file exception happened
```

So it is probably not due to the specific model.
Greetings
Sepp

clsergent, 2023-01-07:
Hi, 
Sorry for the delay... I'm very glad to help you freely!
Indeed, the error is not related to your model. I did generate correct gcode on my computer using it.

Based on your reports, I think that your issue is OS related. As I have no Windows machine, I could not reproduce the error. However, I think there is a problem with how the script restores the selection on Windows.
Can you try this
[custom version](https://github.com/clsergent/Snapmaker2Postprocessor/files/10367100/Snapmaker2_post.custom.py.zip) and report?
Thanks

spitzlbergerj, 2023-01-08:
Good morning,
I ran yesterday with the code generated by your postprocessor (--no-thumbnail) on snapmaker. It all worked out great. The gcode works flawlessly.

I have tried the custom version. The same error appears.

I then messed around with line 114.

First I included the output of a warning that shows me the generated filename.
```
114        FreeCAD.Console.PrintWarning(f'Filename Thumbnail: {file.name}\n')
```

The generated filename looks like this
`Filename Thumbnail: C:\Users\Sepp\AppData\Local\Temp\tmpkp747jw8.png`

Then I changed the new line 115 (old line 114) to

**try 1**
`FreeCADGui.activeDocument().activeView().saveImage("tmp9gyp6iuw", 720, 480, 'Transparent')  # save file`

_**this works**_, the gcode file is generated correctly

**try 2**
`FreeCADGui.activeDocument().activeView().saveImage("C:\Users\Sepp\AppData\Local\Temp\tmp9gyp6iuw", 720, 480, 'Transparent')  # save file`

this does not work, a new error appears. I noticed that the syntax highlighter in Visual Studio Code highlighted the 't' after the last '\'. Therefore I make try 3 without a 't'

error: `(unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape (Snapmaker2_post.py, line 115)`

**try 3**
`FreeCADGui.activeDocument().activeView().saveImage("C:\Users\Sepp\AppData\Local\Temp\mp9gyp6iuw", 720, 480, 'Transparent')  # save file`

same result does not work

**try 4**
FreeCADGui.activeDocument().activeView().saveImage("C:\irgendwas", 720, 480, 'Transparent')  # save file

_**this works**_, the gcode file is generated correctly and after all no file at C:\irgendwas ....

**try 5**
FreeCADGui.activeDocument().activeView().saveImage("C:\irgendwas.png", 720, 480, 'Transparent')  # save file

this does not work, old error  `Unknown file exception happened`

**try 6**
FreeCADGui.activeDocument().activeView().saveImage("D:\irgendwas.png", 720, 480, 'Transparent')  # save file

**_this works!_** and at D:\irgendwas.png there was a correct thumbnail

**try 7**
FreeCADGui.activeDocument().activeView().saveImage("D:\temp\nochwas\wiederwas\undnocheiner\undeinweiterer\irgendwas.png", 720, 480, 'Transparent')  # save file

this does not work, new error

```
post: Snapmaker2(D:/OneDrives/OneDrive - La Gondola Barocca/_PCBüroLustheim/Eigene Dateien/CAD, Designer, Publisher/_CNC-Fräsungen/TestModell CNC/testModell.cnc, )
09:42:42  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 215, in exportObjectsWith
    processor = PostProcessor.load(postname)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPostProcessor.py", line 61, in load
    exec("reload(%s)" % "current_post")
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\bin\lib\importlib\__init__.py", line 169, in reload
    _bootstrap._exec(spec, module)
  File "<frozen importlib._bootstrap>", line 619, in _exec
  File "<frozen importlib._bootstrap_external>", line 879, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1017, in get_code
[... 18 lines cut ...]
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "C:\Program Files\FreeCAD\FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 638, in export
    post.export(objects, filename, argstring)
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 544, in export
    if self.conf.thumbnail and (thumbnail := getThumbnail(self.job)):
  File "C:\Program Files/FreeCAD/FreeCAD_0.20.2-2022-12-27-conda-Windows-x86_64-py310/Mod/Path/PathScripts/post\Snapmaker2_post.py", line 115, in getThumbnail
    FreeCADGui.activeDocument().activeView().saveImage("D:\temp\irgendwas.png", 720, 480, 'Transparent')  # save file

Directory where to save image doesn't exist
```

However, the error message is misleading because the directory definitely exists and is writable for the logged-in Windows user. I had just created it myself!! Below is a screendump of a terminal session where I go into the directory and create a new file with the filename. 

[image]

Writing to the originally determined directory also works, as this screendump shows. Surprisingly there are some files with the correct name structure ...

[image]

Unfortunately, my Python knowledge does not go very far. But it seems to have something to do with the structure of the filename or with the path to the file. It probably has nothing to do with the permissions ....

spitzlbergerj, 2023-01-08:
I hope this helps a bit.
Regards 
Josef

clsergent, 2023-01-09:
Hi,

Thank you for your complete report. I must apology as I did not properly read your comment https://github.com/clsergent/Snapmaker2Postprocessor/issues/1#issuecomment-1373434308, which already indicated the error was related to the file generation (line 119...).

As you experimented yourself, the error seems OS specific and related to the file generation. Several comments on your tests:

1. It works, great! It also indicates that the error is not related to the FreeCAD method `FreeCADGui.activeDocument().activeView().saveImage`, but rather to the file handling.

2. Backslashes are not escaped, which explains why Python complains (`\t` is a tabulation). You can escape the whole string with the raw prefix: `r'C:\Users\Sepp\AppData\Local\Temp\tmp9gyp6iuw'`. You will then notice if you call `print` that all backslashes are escaped. Following tries are more or less affected by this error (n°3, n°5, n°7)

I can't explain however why it works with tries n°4 and n°6...

I am currently installing FreeCAD 0.20.2 on a Windows VM to try to reproduce your setup and your errors. I will get back to you if I find anything

spitzlbergerj, 2023-01-09:
Hi,
no problem and thank you again!

clsergent, 2023-01-22:
Hi,

Sorry for the delay, it took me time to run an x86_64 Windows 10 on my ARM computer. When It finally worked, I found that FreeCAD required 3D acceleration... So I used my 10 years old Mac and VirtualBox, and found the origin of the issue!

It is actually written in the [tempfile](https://docs.python.org/3/library/tempfile.html#tempfile.NamedTemporaryFile) module documentation that on Windows, a temporary file cannot not be opened twice, while it is perfectly valid on Unix (which explains why I never encountered the error).

I just committed a patch (a53181e3a789aa66bbd6165c1d39bb727dd7d943) which should solve your issue. Could you report if it works with your setup?

spitzlbergerj, 2023-01-29:
Hi, 
I have tested the new patch. It works perfectly. I also compared the files created by the previous version of the post processor with the ones created now with and without thumbnail. All files were identical except for the creation date and even the thumbnail.

And one more feedback on the use of the post processor. I have now created a whole series of CNC paths and then milled them on the Snapmaker CNC milling machine. All the milling has been error free and superbly accurate. Thank you very much for this great post processor.

Greetings 
Sepp

closed 2023-01-29 by spitzlbergerj