clsergent/Snapmaker2Postprocessor #2: Include in FreeCAD distribution

PLChris, 2023-01-14:
The topic says it: Would you mind making a pull request to integrate this into the FreeCAD sources?

clsergent, 2023-01-14:
Hi,
Thanks for suggesting this, I will definitely do this when issue #1 will be solved (and thus I consider the script mature enough).

clsergent, 2023-01-24:
An example of pull request to add a postprocessor, including requirements to meet : [https://github.com/FreeCAD/FreeCAD/pull/8193](https://github.com/FreeCAD/FreeCAD/pull/8193)

luzpaz, 2025-01-28:
soft bump. Maybe it's time ?

clsergent, 2025-01-28:
Hi,

Well I would agree but Postprocessor API just changed before release of v1.0
I have a bit of work to comply with new guidelines. In my to-do list but can't say when it will be done!

luzpaz, 2025-01-28:
cc @LarryWoestman just putting this on your radar (if you hear of anyone that needs this post-processor.

LarryWoestman, 2025-01-28:
I have been working on changing the refactored* postprocessors over to the new API, so I am pretty familiar with that process if you have any questions.

The "old-style" API isn't going away any time soon, if you prefer to wait a bit on using the new API.

clsergent, 2025-01-29:
Hi @LarryWoestman,

Thank you for your kind help. I did refactored my code yesterday to fit the new API. 
The API really helps with reusable primitives (drill conversion not to mention it!) which favour code factorisation.
I still had to create a custom `export_common` method to customise the header (including various metadata and a thumbnail), but it is worth it.

However I found UtilsArguments scheme quite complex, with little benefit from my point of view (but maybe I did not explore every user case!).

Indeed, multiple dict are created (values, arguments_visible, argument_defaults), with multiple methods, each of which may have to be reproduced in every postprocessor.

Why not rely solely on argparse by adding extra attributes to generated actions (visible, default help) with a bit of glue:
```
def add_argument(parent:argparse.ArgumentParser, name, *flags, default, visible:bool, help:str, **kwargs) -> argparse.Action:
    default_help = help
    if not visible:
        help = argparse.SUPPRESS
    action = parent.add_argument(*flags, dest=name,default=default, visible=visible, help=help, **kwargs)
    action.default_help = help
    action.visible = visible
    return action
```

This way, extra arguments could be handled the same way as shared ones.
It would even allow to generate help with all arguments no matter their visibility by changing help to default_help value without creating a new parser.

Another way would be to create a custom object definition such as (naive implementation):
```
class Value:
    def __init__(self, name: str, default, visible:bool, help=''):
        self.name = name
        self.default = default
        self.visible = visible
        self.help = help
```

This object could also handle argument generation (and flag arguments with a child class):

```
    def set_argument(self, parent: argparse._ActionsContainer, *args, **kwargs) → argparse.Action :
        if self.visible:
            help = self.help
        else:
            help = argparse.SUPPRESS
        return parent.add_argument(*args, dest=self.name, help=help, default=self.default, **kwargs)
```

Another question without direct relation to my previous comment: argpase has a mechanism to handle flag arguments using `store_true`/`store_false` actions and the same `dest`  attribute. Did something prevented from using this behaviour?

I would see something like:
```
def add_bool_arguments(parent:argparse._ActionsContainer, true_name, false_name,
                       default:bool, visible:bool, true_help:str, false_help:str) -> (argparse.Action, argparse.Action):
    """Create an argument specification for an argument that is a flag."""
    if visible:
        if default:
            true_help = true_help + " (default)"
        else:
            false_help = false_help + " (default)"
    else:
        true_help = false_help = argparse.SUPPRESS
    return (parent.add_argument(true_name, action="store_true", help=true_help, default=default),
            parent.add_argument(false_name, action="store_false", help=false_help, dest=false_name))
```

Anyway that's just ideas, thanks for upgrading postprocessing API!

LarryWoestman, 2025-01-30:
I am working my way (a little bit at a time) to giving postprocessors that are using the shared code (UtilsArguments, UtilsExport and UtilsParse) enough flexibility to be able to be customized "enough" for most purposes.  Right now the "framework" in each of the refactored* postprocessors allows you to add additional arguments and values in ways that appear to me to be very similar to the ways you are describing.    It is quite possible that there are more code-efficient ways to deal with specifying and parsing the arguments.  Right now it is pretty simple but somewhat "brute force".

I still have work to do to allow someone to add to or modify the shared "logic".  I have been requested to do "many" "small" pull requests rather than a smaller number of larger pull requests, so it is taking a while :-)

I also have rewritten the refactored* postprocessors to use the new API and be much more object-oriented.  That code is currently in a pull request and is waiting for review.

An additional goal that I am partway to implementing is to be able to "inherit" from an existing refactored* postprocessor and only have to specify the things that change.  That is also still "under construction".  I plan to have documentation and examples of how to customize postprocessors beyond the command line arguments.

clsergent, 2025-01-30:
Hi,

I think you're perfectly right about inheritance. That's the key to enhance post-processor writing. 

I also understand you can't take every single idea into account.

I'll be following this development with interest. Thank you!

luzpaz, 2025-02-01:
Thanks for considering this ticket @clsergent and thank you for weighing-in @LarryWoestman 🙏

clsergent, 2025-02-07:
I have just finished to adapt the postprocessor today (see new branch [refactor](https://github.com/clsergent/Snapmaker2Postprocessor/tree/refactor)). I have also written a test suite so I think I'm now ready to integrate the postprocessor in FreeCAD. 
However I have not tested it on a real machine so far and I will not be able to do so before two weeks at least. So if you have any time to do so I would be really interested.

LarryWoestman, 2025-02-19:
I don't have access to an appropriate machine, either.

MxRemy, 2025-03-11:
I'm a total novice to all of this stuff, but I have access to an Artisan that I can test it on, if you still need. Yesterday I tried your post processor in FreeCAD 1.0 and didn't run into any errors software-side, but the machine didn't recognize the generated .nc file for some reason. Might've been user error or something obvious though, I'll try again in a few days. Other people encountered this same thing coming from other software (i.e. lightburn), so it might be some simple fix on my end.

clsergent, 2025-03-11:
Hi @MxRemy,
Thanks for your kind help, I'm very interested by your feedback as I don't own an Artisan, just an (quite old now) A350.

Basic check: did you try to change the `.nc` extension to `.cnc`? This trick usually works. I will add it to the wiki.

Another question: did you use the *main* or the *refactor* branch? I'm no longer working on the *main* branch (will be changing branches names soon to reflect this position), just fixing bugs. The *refactor* branch is on its way to be merged into FreeCAD and is supposed to work seamless.

I'd be happy to read about your experiences with the Artisan!

MxRemy, 2025-03-11:
Ooh, I didn't try renaming it and also I was using main! I'll try these next time I'm at the machine, thanks.

I'd definitely be happy to provide any kind of feedback you'd like. This is a real life saver lol, I don't care for Fusion. Once I try it I'll post about the results.

clsergent, 2025-03-13:
Well this is an encoding error: you have supplied an invalid character (accent, umlaut, etc...) in the file name or in the CAM object. If you stick to ASCII it will not happen

By the way, I will close this issue as the post processor has been merged. Please open a new issue with this encoding error, I will look at it tomorrow.

MxRemy, 2025-03-13:
Will do! That's awesome

clsergent, 2025-03-13:
The post processor has been merged today in FreeCAD (#PR20154).