clsergent/Snapmaker2Postprocessor #3: Multiplatform Support

ITmaze, 2023-01-15:
I'm raising this as an issue, but really it's a place to have a discussion. The prompt was a log entry in FreeCAD during PostProcessing: "Command ignored: G83", which turned out to be a drilling operation which was most of the job at hand. I understand that SnapMaker (and Marlin) G-Code does not support G83 and that's fine, but it raised this issue.

I was using my A350 in CNC "mode", but I also want to be able to post process Laser and 3D printing from within FreeCAD.

I understand that each mode has specific requirements and I wondered if there is value in creating three separate post processing modules, one for each mode. Perhaps it should be per modules, differentiating between the 1600 mW and 10 W laser and the Single vs Dual-Extruder, perhaps take into account the 4th axis rotary module, etc.

The point being that making a single Post Processor that handles all this seems less than ideal.

In addition, there is no reason that I can think that a G-Code like G83 can't be replaced by a few lines of G-Code and thus adding G83 support to the hardware. Note that G83 is just an example. You're alreading replacing unsupported tool changes with a pause, so this idea is already happening.

I understand that this is a non-trivial process. I'm a 40-year software development professional, but my journey into SnapMaker and G-Code is still very young.

I realise that "patches welcome" is a perfectly valid response, but I'm hoping to learn if this is worthwhile before digging in too deep.

Kind regards,
Onno

clsergent, 2023-01-15:
Hi,

As your post actually raised both a discussion, and a potential issue, I will try to answer to both singly:

## G83:

The script manages `G83` by converting it into Snapmaker gcode (multiples `G0`and `G1` actually). The process is handled by the `Postprocessor.translateDrill` method.

You got the `Command ignored: G83` because of a minor error in the code. The command was nonetheless processed. I just committed cd2d0c4f9a858f4dbe36cc263b61b68d6809a71e which should solve this issue. You should not get the warning again (unless you pass `--no-translate-drill-cycles` of course). Could you confirm it?

## Discussion: Evolution of the postprocessor

I agree on the fact that it would be ideal to be able to use FreeCAD for every operation on the Snapmaker. I made this postprocessor as I did not find any software matching my expectations for CNC operations (and I use FreeCAD for 3D modelling).

I would be interested in doing the same for laser operations, and it should not be so had to do another postprocessor (or add a laser mode). I don't know however if the Path workbench is fitted for laser operations. I will take a look when I can find a bit of time.

As for 3D printing operations, I am currently using Cura, which engine is very mature, so I will not attempt anything in this direction, but as you mentioned, "patches welcome"!

ITmaze, 2023-01-19:
I've not responded because I'm unsure how to report what I'm seeing, since it's unclear if it's caused by FreeCAD or the post processor. I'm seeing loads of

clsergent, 2023-01-19:
Hi,

I have received your answer but it seems truncated… I would really appreciate if you could report what you got.
Thanks!

ITmaze, 2023-01-19:
For reasons I don't understand GitHub submitted my incomplete report just before my workstation crashed. I deleted it, but by then you had been notified, sorry for the noise.

Thank you for your comments and fix. Perhaps we should separate the issues? I propose that we keep this one as the "G83 drilling issue" and open a new one for the multiplatform one. Thoughts?

The G83 warning no longer occurs, but now I can see a whole bunch of `G0: negative or null speed provided (0.0)` and `G1: negative or null speed provided (0.0)` notices which result in camotics reporting:

```
ERROR:Cutting move with zero feed
ERROR:While executing GCode block:G1 Z16 F0
ERROR:       At: {redacted}.cnc:27
ERROR:Caused by: Cutting move with zero feed.
```

I also see an `Command ignored: M6`, which I thought were translated into a pause.

I think that the `Command ignored: G80` could be a left over from implementing the G83 code in a different way.

clsergent, 2023-01-20:
Hi,

Thank you for reporting, it helps me improving the script! 

I agree with your proposal to split the initial issue. I have thus opened a new issue (#4) to discuss about Multiplatform support. Below is my answer to the issues you just reported:

## `negative or null speed provided` warning
This warning occurs when no speed has been defined for the selected Job in FreeCAD. You can set up in the Job *Tool* tab:
- rapid speed used by `G0`:
  [image]

- tool speeds used by `G1`:
  [image]

## `Command ignored: M6`
Snapmaker gcode does not include `M6` (tool change). The postprocessor adds an HMI pause (`M76`) but also warns about the removal of `M6` commands. As you are not the first one to be confused by this behaviour (see https://github.com/clsergent/Snapmaker2Postprocessor/issues/1#issuecomment-1372528666), I may remove this warning in the future.

## `Command ignored: G80`
`G80` (cancel current motion mode) was not present in the Snapmaker GCODE [documentation](https://snapmaker.github.io/Documentation/gcode/G000-G001), which explains why the script ignored it. However the last firmware explicitly refers to this command, so I will include it in the list of supported commands. Thanks for pointing this out.

As every drilling operation (`G81...G83`) is explicitly written by FreeCAD, it doesn't seem necessary to modify the drilling translation method but I may be wrong...

Could you confirm if my answer is sufficient or if I need to change anything else?

ITmaze, 2023-01-21:
It appears that this patch fixes the reported problem, thank you.

After entering the Rapid and Tool speeds as suggested, all the `G0` and `G1` errors went away. During testing I discovered a new issue #5.

I note that the context and clarification that the two images shown in the discussion of this issue made me wonder if it would be helpful to create an FAQ that could be referred to in the error message, so a new user - truth be told, we're all one of those at one time or another - could resolve the issue using information that's documented in perhaps a wiki page? I'd be happy to contribute.

clsergent, 2023-01-21:
Glad to know that the problem has been fixed!
You're right about the wiki page. I will begin a FAQ with good practices and common errors/mistakes.

closed 2023-01-21 by clsergent