clsergent/Snapmaker2Postprocessor #5: Crash when model is supplied on FreeCAD command line

ITmaze, 2023-01-21:
In trouble-shooting #3  I discovered that if you supply the filename on the command line to launch FreeCAD (`freecad top_plate.FCStd`), the post processor crashes with the following error:

```
11:37:36  post: SnapMaker_A350_CNC(/home/onno/Documents/20230117.mounting-plate/top_plate.cnc, --no-thumbnail)
11:37:36  usage: snapmaker2_post [-h] [--header] [--no-header] [--comments] [--no-comments] [--comment-symbols COMMENT_SYMBOLS COMMENT_SYMBOLS] [--thumbnail]
                       [--no-thumbnail] [--line-numbers] [--no-line-numbers] [--line-start LINE_START] [--line-increment LINE_INCREMENT] [--remove-duplicates]
                       [--keep-duplicates] [--show-editor] [--no-show-editor] [--precision PRECISION] [--pause {M25,M76}] [--units {mm,in}] [--preamble PREAMBLE]
                       [--postamble POSTAMBLE] [--pre-operation PRE_OPERATION] [--post-operation POST_OPERATION] [--translate-drill-cycles]
                       [--no-translate-drill-cycles] [--tool-change [TOOL_CHANGE]] [--no-tool-change] [--tool-number] [--no-tool-number]
                       [--spindle-wait SPINDLE_WAIT] [--spacer SPACER] [--commands COMMANDS [COMMANDS ...]] [--final-position FINAL_POSITION]
11:37:36  snapmaker2_post: error: unrecognized arguments: top_plate.FCStd
11:37:36  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "/tmp/.mount_freecavrWwC4/usr/Mod/Path/PathScripts/PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "/tmp/.mount_freecavrWwC4/usr/Mod/Path/PathScripts/PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "/tmp/.mount_freecavrWwC4/usr/Mod/Path/PathScripts/PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "/home/onno/.FreeCAD/Macro/SnapMaker_A350_CNC_post.py", line 641, in export
    post = Postprocessor()
  File "/home/onno/.FreeCAD/Macro/SnapMaker_A350_CNC_post.py", line 306, in __init__
    self.configure()
  File "/home/onno/.FreeCAD/Macro/SnapMaker_A350_CNC_post.py", line 385, in configure
    self.conf = parser.parse_args(*args)
  File "/tmp/.mount_freecavrWwC4/usr/lib/python3.10/argparse.py", line 1829, in parse_args
    self.error(msg % ' '.join(argv))
  File "/tmp/.mount_freecavrWwC4/usr/lib/python3.10/argparse.py", line 2583, in error
    self.exit(2, _('%(prog)s: error: %(message)s\n') % args)
  File "/tmp/.mount_freecavrWwC4/usr/lib/python3.10/argparse.py", line 2570, in exit
    _sys.exit(status)

2
```

This error does not occur if the file is opened from within FreeCAD. The output file is set to `top_plate.cnc`.

Note that I've renamed the post processor on my system to `SnapMaker_A350_CNC_post.py`, this is the same file as `Snapmaker2_post.py`.

clsergent, 2023-01-21:
I have been able to reproduce your issue, which actually happens whatever argument supplied.

For any reason `argparse.ArgumentParser.parse_args()` method keeps arguments passed to FreeCAD at startup when called by the postprocessor.

I committed (1163cd88e52f7ec0941501388035fae0e3679cb9) to fix this issue by explicitly passing arguments to the `parse_args()` method. 

Could you report if it solves the issue?

ITmaze, 2023-01-23:
I can confirm that this now works as expected.

closed 2023-01-23 by ITmaze