clsergent/Snapmaker2Postprocessor #6: Feature Request: Boundary Checking

ITmaze, 2023-01-25:
Whilst post processing the G-Code from FreeCAD it should be possible to track the movements in the various axes to determine if they exceed the physical dimensions of the Snapmaker for which the G-Code is intended.

Things that will need to be considered:
- There's likely no origin defined
- Each Snapmaker model has different dimensions
- The stops on the device are soft stops by default, but they can be disabled for a larger work area
- It probably needs to take the form of an alert rather than an error, along the lines of: "It appears that your model exceeds the physical boundaries of your device."
- It could probably accumulate moves and store them in terns of min and max for each axis. If the difference between these exceeds the device extent, issue a warning.
- Bezier and other curves will need to be accommodated, since even if Snapmaker G-Code doesn't support it today, [it can be enabled](https://forum.snapmaker.com/t/g2-g3-and-snapmaker-2/13798/4) and might be standard at some point.
- Boundaries might also be configurable to accommodate setups with different requirements.

clsergent, 2023-01-25:
I created a new branch to test the boundaries check implementation. 
It currently only checks G0/G1 moves (and thus converted G81-G83), and issue a warning if physical dimensions are exceeded. Other moves will be added later.

Instructions:
- add `--machine=<machine>` and `--boundaries-check` to activate boundaries check.
- `--machine` arg may be omitted if `MACHINE` constant is set up
- check help to get machine names, or look at `BOUNDARIES`constant

clsergent, 2023-01-30:
The post-processor is indeed not designed to be run standalone as it requires FreeCAD GUI (even though I try to remove this dependancy). 

You can pass `--help` as a postprocessor argument but I agree that it is inelegant... 
I just pushed a commit to change the post-processor behaviour to print help when run from the command-line. You can run it as a macro to print help.

I also added an argument (`--boundaries <X,Y,Z>`) to specify custom boundaries. If provided, it overrides `--machine`.

ITmaze, 2023-01-31:
Thank you, I'll add this to my list to test.

ITmaze, 2023-02-10:
Thanks for your patience, had to rebuild my CNC VM.

Testing gives me some unexpected issues. I've documented as they occurred. Not sure if this is complete.

## Test 1 - Running on the command-line with `--help`

There is no help returned and the output appears to be the same as previously, though I've not actually compared it.

    $./Snapmaker_2_CNC_post.py --help
    ./Snapmaker_2_CNC_post.py: line 3: import: command not found
    ./Snapmaker_2_CNC_post.py: line 4: import: command not found
    ./Snapmaker_2_CNC_post.py: line 5: import: command not found
    ./Snapmaker_2_CNC_post.py: line 6: import: command not found
    ./Snapmaker_2_CNC_post.py: line 7: import: command not found
    ./Snapmaker_2_CNC_post.py: line 8: from: command not found
    ./Snapmaker_2_CNC_post.py: line 9: import: command not found
    ./Snapmaker_2_CNC_post.py: line 10: import: command not found
    ./Snapmaker_2_CNC_post.py: line 12: import: command not found
    ./Snapmaker_2_CNC_post.py: line 13: import: command not found
    ./Snapmaker_2_CNC_post.py: line 14: import: command not found
    ./Snapmaker_2_CNC_post.py: line 15: import: command not found
    ./Snapmaker_2_CNC_post.py: line 16: import: command not found
    ./Snapmaker_2_CNC_post.py: line 18: __version__: command not found
    ./Snapmaker_2_CNC_post.py: line 19: __author__: command not found
    ./Snapmaker_2_CNC_post.py: line 20: __license__: command not found
    ./Snapmaker_2_CNC_post.py: line 23: GCODE_MOTION_MODE: command not found
    ./Snapmaker_2_CNC_post.py: line 24: GCODE_WORK_PLANE: command not found
    ./Snapmaker_2_CNC_post.py: line 28: MACHINE: command not found
    ./Snapmaker_2_CNC_post.py: line 29: UNITS: command not found
    ./Snapmaker_2_CNC_post.py: line 30: PRECISION: command not found
    ./Snapmaker_2_CNC_post.py: line 31: MAX_SPINDLE_SPEED: command not found
    ./Snapmaker_2_CNC_post.py: line 32: MIN_SPINDLE_SPEED: command not found
    ./Snapmaker_2_CNC_post.py: line 33: TRANSLATE_DRILL_CYCLES: command not found
    ./Snapmaker_2_CNC_post.py: line 34: DRILL_RETRACT_MODE: command not found
    ./Snapmaker_2_CNC_post.py: line 35: TOOL_CHANGE: command not found
    ./Snapmaker_2_CNC_post.py: line 36: SPINDLE_WAIT: command not found
    ./Snapmaker_2_CNC_post.py: line 37: PAUSE: command not found
    ./Snapmaker_2_CNC_post.py: line 38: REMOVE_DUPLICATES: command not found
    ./Snapmaker_2_CNC_post.py: line 39: LINE_START: command not found
    ./Snapmaker_2_CNC_post.py: line 40: LINE_INCREMENT: command not found
    ./Snapmaker_2_CNC_post.py: line 41: BOUNDARIES_CHECK: command not found
    ./Snapmaker_2_CNC_post.py: line 44: INCLUDE_HEADER: command not found
    ./Snapmaker_2_CNC_post.py: line 45: INCLUDE_THUMBNAIL: command not found
    ./Snapmaker_2_CNC_post.py: line 46: INCLUDE_COMMENTS: command not found
    ./Snapmaker_2_CNC_post.py: line 47: INCLUDE_LINE_NUMBERS: command not found
    ./Snapmaker_2_CNC_post.py: line 48: INCLUDE_TOOL_NUMBER: command not found
    ./Snapmaker_2_CNC_post.py: line 53: syntax error near unexpected token `('
    ./Snapmaker_2_CNC_post.py: line 53: `BOUNDARIES = dict(original=dict(X=90, Y=90, Z=50),'

## Test 2 - Using FreeCAD post processor parameters: `--machine=A350 --no-thumbnail --boundaries-check`

Results in a CNC file with a thumbnail and this warning on the console:

    12:06:52  Boundary check failed, no valid machine name supplied

## Test 3 - Using FreeCAD post processor parameters: `--boundaries-check --boundaries=90,90,50`

Gives the following warning:

    12:17:08  Boundary check failed, no valid machine name supplied

clsergent, 2023-02-10:
Hi!

## Test 1

I did not include a shebang at the top of the script. Thus you must call explicitly python to run the script (ie. `python ./Snapmaker_2_CNC_post.py --help`). Note that FreeCAD modules must be in your PYTHONPATH.

However, I was referring to adding `--help` in the "Post Processor Args" box of the Job settings. Alternately, you can call the post processor as a macro without any argument, which is more convenient. Both ways will output to the FreeCAD report view.

## Test 2 and 3
I tried both without any issue. Did you use last commit version of the 6-feature-request-boundary-checking branch? Could you provide your configuration (OS, FreeCAD and Python version, etc.) and check that all messages types are checked in the report view 
    [image]

ITmaze, 2023-02-10:
## Versions:

* AppImage, FreeCAD 0.20.2, Libs: 0.20.2R29603 (Git)
* Debian GNU/Linux 11 (bullseye)
* Snapmaker2Postprocessor commit: 04f4e788746068778dd65a47f8744b3f2a0fec81

I removed the `__pycache__` directory to ensure that I wasn't running a cached version of the post processor.

## Test 1: Running on the command-line with `--help`

When I run it using `python3 Snapmaker2Postprocessor`, the code cannot load FreeCAD, not surprising, since the AppImage isn't on the path, nor do I expect it to have loadable modules. The import for FreeCAD doesn't have to happen before processing the `--help` argument, so help could reasonably be processed before the main code executes.

    ModuleNotFoundError: No module named 'FreeCAD'

Noted on the shebang. If I add it, FreeCAD doesn't seem to care and it allows you to run it from the CLI.

Running it as a macro, which I hadn't until just now, I can finally see the help. Excellent.

## Test 2: Using FreeCAD post processor parameters: `--machine=A350 --no-thumbnail --boundaries-check`

When I turn on all logging messages the output is:

    07:40:01  failed to create compose table
    07:40:04  post: Snapmaker_2_CNC(/home/onno/Documents/20230117.mounting-plate/2023021101.top_plate.cnc, )
    07:40:04  Post Processor: Snapmaker_2_CNC_post
    Postprocessing...
    07:40:04  No parent job was found for <PathScripts.PathPost._TempObject object at 0x7f913a1eaa40>
    07:40:04  No parent job was found for <Path::FeaturePython object>
    07:40:04  Processing object Fixture
    07:40:04  Processing object TC__4mm_Drill002
    07:40:04  Processing object Drilling
    07:40:04  Postprocessing done
    07:40:04  Boundaries check/nBoundary check failed, no valid machine name supplied

clsergent, 2023-02-24:
Hi,

Sorry for the delay... I have been away for a while

## Test 1:
### shebang
I did not add a shebang as the Python path is not consistent across OS, and the script is not intended to be used standalone.
However I found that [PEP 394](https://peps.python.org/pep-0394/#recommendation) actually recommends `#!/usr/bin/env python` so I will add this.

### Conditional import of FreeCAD modules
I am not in favour of conditional import for two main reasons:
- it adds script lines/statements with little benefit as the script is useless without these modules (except for `--help`, I admit) and the less the better;
- it prevents from using type hints, which I use to easily document the script. It s true that type annotations can be stringified to avoid errors but it is inelegant.

I would like to know your opinion about this before changing anything.

## Test 2:
Thanks for posting your log. It shows that your arguments are not taken into account. Indeed, you got

> 07:40:04  post: Snapmaker_2_CNC(/home/onno/Documents/20230117.mounting-plate/2023021101.top_plate.cnc, )

While you should have got (with arguments):
 ```07:40:04  post: Snapmaker_2_CNC(/home/onno/Documents/20230117.mounting-plate/2023021101.top_plate.cnc, --machine=A350 --no-thumbnail --boundaries-check)```
Did you write the arguments at the right place? They should appear here :
[image]

Otherwise it means that the Path Workbench fails to pass arguments to the postprocessor... It may

vk6flab, 2023-02-27:
Noted on the shebang.

I'll have a think about the conditional import of modules and do some tests. I'll get back to you on this.

As for test 2, the plot thickens. I had put the parameters as the "Default Arguments" in the Post Processor tab of the Path Preferences pane. It did not appear to use those. When I put them into the setup sheet for the job it took note of the settings. I haven't yet actually tested the arguments, but just to report that they are now being "seen".

clsergent, 2023-02-27:
Actually "Default Arguments" are only added to a new job, never to an already existing one, as they could override a custom value. If you create a job from scratch, you should see default arguments in the "Post Processor Args" section of the job.

ITmaze, 2023-02-27:
I see a "Warning: no parameters specified" message and a matching FAQ in
our future :)

It's also interesting that I changed the parameters in the preferences
pane, but don't recall any feedback that they would be ignored for existing
jobs.
--
finger painting on glass is an inexact art - apologies for any errors in
this scra^Hibble

()/)/)() ..ASCII for Onno..

ITmaze, 2023-03-13:
Finally had a moment to test. Thank you for your patience. In short, it appears to work as advertised for me.

- The boundary check warning doesn't appear in any other location than the log/console, which means that many people will likely never see it. It appears that there is a way to raise an alert from within a macro using a MessageBox[^1] - The code has currently got many ``FreeCAD.Console`` calls, perhaps they should be abstracted to emit both a console message and a messagebox as needed.
- What do you think about inserting the boundary warning into the G-code as a comment?
- Using ``--no-boundary-check`` works as expected, but I think it would be helpful if it said so, since it's specified as an argument.
- The *Snapmaker-Artisan* has a boundary of 400x400x400 by the looks of the specs[^2] - I note that the page doesn't specify anything other than 400x400 for the CNC module, but specifies 400x400x400 for 3D printing. I realise that the end-mill and material used will likely determine final maximum height possible, so perhaps Snapmaker's reasoning for not specifying height is based on that.
- The last line on the script never gets executed on my system, ``Postprocessor().configure('--help')``, since it fails when FreeCAD fails to import. I've been looking into conditional import and type checking. There are ways to achieve this[^3]. Using ``sys.executable.find("freecad")``[^4][^5], you can discover if it's running inside FreeCAD. Combining this should make it so that the imports could be conditional. I note that even if you comment out the imports for ``FreeCAD``, ``Path`` and ``PathScripts``, the code still doesn't show ``--help``, it fails with ``def getSelectedJob() -> PathJob.ObjectJob:``, indicating that I suspect there's a structural error in relation to how help is shown. I wonder if running the help module from the shebang and having a conditional import is the way to go.

Note that I'm acutely aware that I've given you a list of problems, but I'm also finally at the point where I can do some more meaningful coding for this, so I plan to give you some patches to go with my problem reports :-)

Also, @vk6flab is also me, but I suspect that you already figured that out, it's my amateur radio callsign.

[^1]: https://wiki.freecad.org/Macro_MessageBox
[^2]: https://snapmaker.com/snapmaker-artisan/specs
[^3]: https://stackoverflow.com/questions/61384752/how-to-type-hint-with-an-optional-import
[^4]: https://stackoverflow.com/a/17755798
[^5]: https://docs.python.org/2/library/sys.html#sys.executable

ITmaze, 2023-03-13:
Created artisan patch: [e57f120](https://github.com/ITmaze/Snapmaker2Postprocessor/pull/1/commits/e57f120b5e54b3c1cfca745c72bb24a2cefbcc64)

ITmaze, 2023-03-14:
I've created a pull request #7 for the Artisan. Sorry for the noise in getting here.

ITmaze, 2023-03-14:
Random idea.

What about adding the ``--help`` output to the README.md file, so we can bypass all the script execution stuff, including any shebang, since that is looking less and less trivial or maintainable.

ITmaze, 2023-03-14:
Created pull request #8 to update the `README.md` to show the output of `--help`.

clsergent, 2023-03-16:
I merged your PR #7. I was not sure if Artisan was processing the same GCode, but it actually does as it is referred to in Luban.

clsergent, 2023-03-31:
Additions to the script:
- conditional import of FreeCAD: allows to get help by running the script from commandline.
- boundary check warnings are now added as GCODE comments (unless --no-comment is provided)
- various bug fixes

I tried various methods to display errors/warnings:
1. Use Qt `QMessageBox`
 \+ catchy
 \- modal windows delay the process
 \- multiple errors lead to multiple windows, which is painful

2. insert a `QLabel` in the `PostUtils.GCodeEditorDialog`
 \+ sums up errors and warnings on top of the GCODE
 \- not very aesthetic (at this stage)
 \- layout may have to be adapted as multiple lines may hide GCODE editor

It looked like : 
   [image]
using this code (as a reminder):
```
messages = [f'Error: {e}' for e in Logger.errors] + [f'Warning: {e}' for e in Logger.warnings]
dialog.label = QtWidgets.QLabel('\n'.join(messages), dialog)
dialog.label.setFrameStyle(QtWidgets.QFrame.StyledPanel | QtWidgets.QFrame.Plain)
dialog.layout().insertWidget(0, dialog.label, stretch=0)
```

In both cases, I had to catch FreeCAD.Console output, which turned out to be quite complex. Indeed, this module is written in C++, which means that standard stdout/stderr redirection fails.

3. Stick to *Report View*
*Report View* is the standard place to display logs, so I decided to stick to FreeCAD guidelines. It turns out that the Report View can become visible when certain logs occur. I guess it's good enough. I added a final generic text that also takes place in the *Status bar*, stating whether the boundary check succeeded or failed. 

How it looks like :
   [image]

As the boundary check features seems mature enough, I will merge  * 6-feature-request-boundary-checking*, and add an entry in the wiki.

Let me know if it fits your needs.

ITmaze, 2023-03-31:
Nice!

I have been thinking about how to solve this and came up with adding the errors to an array and then presenting them all in a final dialog box, but your discovery of the FreeCAD guidelines supersedes that.

Happy to close this, thank you for your coding!

clsergent, 2023-04-11:
Glad to help!
I close the issue, but the enhancement of the boundary check function is still on my roadmap (especially for arc movements).

closed 2023-04-11 by clsergent