clsergent/Snapmaker2Postprocessor #9: Cannot import modules on 0.21

rakhusur, 2023-08-26:
Hi,
thanks for making this post! I tried using it on 0.21, but I get
`FreeCAD modules could not be imported. Only help is available`.

clsergent, 2023-08-27:
Hi,

Could you provide your OS and how you tried to run the postprocessor to help me understand the context of execution?

From what you encountered, it seems that the postprocessor has been run standalone and that FreeCAD modules are not in `sys.path`.

The usual way to use the postprocessor is through the GUI (please read this section: https://github.com/clsergent/Snapmaker2Postprocessor#usage). 

If you tried to use it on command line, I advise you to run the postprocessor from the FreeCAD console, as FreeCAD modules are accessible there.
Otherwise, you must find where FreeCAD python modules are located (which is OS-dependant) and add them to PYTHONPATH.

rakhusur, 2023-08-27:
I'm on arch linux using FreeCAD from the official repos, which is version 0.21.0. I was setting the post via the GUI (the message above appeared in the report view). I've been digging a bit through some scripts this morning and it looks to me like there was an API change: if I replace
`
    import PathScripts.PathUtil as PathUtil
    import PathScripts.PostUtils as PostUtils
    import PathScripts.PathJob as PathJob
`
by
`
    import Path.Base.Util as PathUtil
    import Path.Post.Utils as PostUtils
    import Path.Main.Job as PathJob
`
it works!

clsergent, 2023-08-27:
I downloaded FreeCAD 0.21 and I can confirm this change. I will update the script to reflect this change. Thank you for your help!

closed 2023-08-27 by clsergent