clsergent/Snapmaker2Postprocessor #10: Problem with hanging freecad build 0.21.2 (linux)

JPDuffy1993, 2024-01-02:
Hi, I have been running into an issue with this on freecad build 0.21.2 where the system hangs when using the post processor, i have tried to diagnose it myself but unfortunately there is no output on the python console in freecad or the freecad console itself, is this something you can look into? 

it looks like the system is trying to do something as i can see CPU and memory allocation so i'm not if its stuck in a loop, also im not a freecad pro so maybe its something im doing incorrectly, do i need to add further arguements to the post processor in freecad? i have seen in the file you have one for the artisan which is what i have so i set that as the machine

Thank you

clsergent, 2024-01-18:
I found time to test the post processor in a Linux Environment, without issue so far. However, it appears that there are many alternative versions of FreeCAD. Could you give me your setup to help me investigate?
I would need at least:
- Kernel/OS
- FreCAD version (including distribution mode such  as Flatpack/AppImage/Snap)
- Python version

You could also try the following actions:
- run FreeCAD from a Terminal to see stdin/stderr (if not done already)?
- import the post processor from the FreeCAD python console itself to see if it raise any error.

JPDuffy1993, 2024-01-22:
Hi Thank you for looking into this too 

I am using Ubuntu 23.10, i tried with FreeCAD 0.21.2 and the latest 0.22 releases, python V3 

I have found importing the post processor directly using Python to be successful so thank you for that!

clsergent, 2024-01-24:
Hi,

I may be mistaken  it seems that you solved your issue? If so, great! Could you then close this issue.

If not, could you provide me a few more data:
- FreeCAD distribution mode (apt / snap /AppImage / FlatPack / custom build...)
- python version (including minor number) used by FreeCAD, which may be different from the system one
- results of stdout/err when poste-processing from FreeCAD itself

Thank you