clsergent/Snapmaker2Postprocessor #11: Errors as soon as I select this postprocessor

spitzlbergerj, 2024-01-06:
Hi,
I have already used this postprocessor successfully in an earlier version of FreeCAD and was very satisfied. After some FreeCAD updates I wanted to create the next CNC job for Snapmaker. Unfortunately, I am now getting error messages again.

**FreeCAD installation:**
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60n
  * Help 1.0.3
  * InventorLoader 1.3.0
  * LCInterlocking
  * sheetmetal 0.2.63

Postprocessor installation:

- current version of Snapmaker_2_CNC_post.py copied to the directory 
`C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\scripts`
- Started FreeCAD 0.21.1
- Opened file with some ShapeStrings
- Path Workbench called up
- New job created, Shapestrings selected
- Changed output to this postprocessor
- 
Then I get the following errors

`
09:30:13  Traceback (most recent call last):
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Main\Job.py", line 552, in onChanged
    processor = PostProcessor.load(obj.PostProcessor)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\Processor.py", line 47, in load
    exec("import %s as current_post" % postname, namespace)
  File "<string>", line 1, in <module>
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users/Sepp/AppData/Local/Programs/FreeCAD 0.21/Mod/Path/Path/Post/scripts\Snapmaker_2_CNC_post.py", line 197, in <module>
    class Command:
  File "C:\Users/Sepp/AppData/Local/Programs/FreeCAD 0.21/Mod/Path/Path/Post/scripts\Snapmaker_2_CNC_post.py", line 216, in Command
    def addParameter(self, parameter, value: str | int | float = ''):
<class 'TypeError'>: unsupported operand type(s) for |: 'type' and 'type'
`

clsergent, 2024-01-15:
Hi!
I am very busy at work at the moment. I may be a bit long to answer but I will try my best to handle it. 
 Thanks!

clsergent, 2024-01-15:
I guess I should have read your post in the first place. It seems that your Python version doesn't handle type annotations. Could you provide me the Python version number used by FreeCAD?

spitzlbergerj, 2024-01-15:
Hi,
Thank you very much for your help. If it takes longer, then so be it. I'm glad that you're taking your time at all.

The Python version is already listed above in the FreeCAD version overview.
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3

when I check the Python version under Windows, it is V 3.10.8, which is a higher version.

I have just installed the latest version of FreeCAD (official download for Windows 64 bit). The Python version included in the bundle is the same as in 0.21.1:
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60n
  * Help 1.0.3
  * InventorLoader 1.3.0
  * LCInterlocking
  * sheetmetal 0.2.63
[/code]

I'm not sure if I can change the Python version of the bundle

clsergent, 2024-01-18:
I just committed to handle your Python version. Actually Python < 3.10 doesn't handle pipe syntax for unions in type annotations. It should fix your error.

spitzlbergerj, 2024-01-19:
Thank you very much. The errors no longer appear.

closed 2024-01-19 by spitzlbergerj