clsergent/Snapmaker2Postprocessor #13: Error in parse_args after starting the postprocessor - FreeCAD 0.21.2

spitzlbergerj, 2024-03-29:
Hello!
I wanted to use the postprocessor again with the current freeCAD version 0.21.2. Unfortunately I get an error message again

**FreeCAD installation:**

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60n
  * Help 1.0.3
  * InventorLoader 1.3.0
  * LCInterlocking
  * sheetmetal 0.2.63

```

**my Path workflow**
```
- current version of Snapmaker_2_CNC_post.py copied to the directory 
`C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\scripts`
- Started FreeCAD 0.21.2
- Opened file with some ShapeStrings
- Path Workbench called up
- New job created with one Shapestrings selected
- Changed output to this postprocessor with option --no-thumbnail
```

Then I get the following errors

```
post: Snapmaker_2_CNC(D:\OneDrives\OneDrive - La Gondola Barocca\_PCBüroLustheim\Eigene Dateien\CAD, Designer, Publisher.nc, --no-thumbnail)
09:47:57  Post Processor: Snapmaker_2_CNC_post
09:47:57  Postprocessing...
09:47:57  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\Command.py", line 583, in Activated
    result, gcode, name = self.exportObjectsWith(sublist, partname, job, idx)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\Command.py", line 518, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\Mod\Path\Path\Post\Processor.py", line 80, in export
    return self.script.export(obj, filename, args)
  File "C:\Users/Sepp/AppData/Local/Programs/FreeCAD 0.21/Mod/Path/Path/Post/scripts\Snapmaker_2_CNC_post.py", line 749, in export
    return post.export(objects, filename, argstring)
  File "C:\Users/Sepp/AppData/Local/Programs/FreeCAD 0.21/Mod/Path/Path/Post/scripts\Snapmaker_2_CNC_post.py", line 625, in export
    self.configure(*shlex.split(argstring))
  File "C:\Users/Sepp/AppData/Local/Programs/FreeCAD 0.21/Mod/Path/Path/Post/scripts\Snapmaker_2_CNC_post.py", line 437, in configure
    self.conf = parser.parse_args(args=args)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 1874, in take_action
    action(self, namespace, argument_values, option_string)
  File "C:\Users\Sepp\AppData\Local\Programs\FreeCAD 0.21\bin\lib\argparse.py", line 1045, in __call__
  
0
```

spitzlbergerj, 2024-03-29:
Hi again,
I don't know why, but just now the postprocessor worked normally without errors.
What did I do? 

- After the errors described above, switched to another postprocessor (which produced even more errors) 
- Switched back to this processor
- selected the job
- let it update once,
- Started Post Processing again
- Result:

10:03:08 post: Snapmaker_2_CNC(D:\OneDrives\OneDrive - La Gondola Barocca\_PCBüroLustheim\Own Files\CAD, Designer, Publisher.nc, )
10:03:08 Post Processor: Snapmaker_2_CNC_post
10:03:08 Postprocessing...
10:03:08 Postprocessing done

I can no longer reproduce the error from above. Even after a new start everything works as desired.

Sorry for opening an issue that is no longer an issue ....

Regards
Sepp

closed 2024-03-29 by spitzlbergerj

clsergent, 2024-03-29:
Hi,
It seems that for any reason a bad argument was provided to the post processor, leading to the error you reported. It also explains why the other post processor failed as well.
I will add exception handling to avoid this particular error. Thanks for reporting!

spitzlbergerj, 2024-03-29:
Thank you so much for maintaining this post processor so well. Thank you!