clsergent/Snapmaker2Postprocessor #14: Error G0: negative or null speed provided (0.0)

datamystic, 2024-06-16:
I am an absolute beginner with FreeCAD and have just started trying my hand at generate variable depth CNC files instead of using SnapMaker 2 Luban, which only does a single depth.
[test-freecad.zip](https://github.com/user-attachments/files/15857560/test-freecad.zip)

I have a basic hexagon prism with radial chamfered edges (file attached). I have added the snapmaker_freecad_post.py to my Macro folder and selected this in the Output options. I have not changed any tools/bits yet as the snapmaker tool .json file seems incompatible with the .fctb file used by other ToolBits

Testing with

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]

I am getting the following errors in the log:

[code]
21:14:52  post: snapmaker_freecad(C:/Users/simon/Downloads/test.gcode, )
21:14:52  Post Processor: snapmaker_freecad_post
21:14:52  Postprocessing...
21:14:52  Postprocessing done
21:14:52  G0: negative or null speed provided (0.0)
[/code]

with no output file generated. Any help appreciated!

clsergent, 2024-06-16:
Hi,

Did you read the [FAQ section](https://github.com/clsergent/Snapmaker2Postprocessor/wiki/FAQ#i-encountered-this-warning-g0-negative-or-null-speed-provided-00) of the wiki about G0 error? You must setup horizontal and vertical rapid speeds (when the CNC toolhead is not working).

Let me know if it solves your problem!

datamystic, 2024-06-16:
Thank you - I hadn't see it. Now I get:

[code]
22:29:42  post: KineticNCBeamicon2(C:/Users/simon/Downloads/test.gcode, )
22:29:42  Show editor = 1
22:29:42  postprocessing...
22:29:42  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.20\Mod\Path\PathScripts\PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "C:\Program Files\FreeCAD 0.20\Mod\Path\PathScripts\PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "C:\Program Files\FreeCAD 0.20\Mod\Path\PathScripts\PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "C:\Program Files/FreeCAD 0.20/Mod/Path/PathScripts/post\KineticNCBeamicon2_post.py", line 227, in export
    job = PathUtils.findParentJob(obj)
  File "C:\Program Files\FreeCAD 0.20\Mod\Path\PathScripts\PathUtils.py", line 405, in findParentJob
    for i in obj.InList:

'_TempObject' object has no attribute 'InList'
[/code]

I tried setting the output path as per:
  https://forum.freecad.org/viewtopic.php?t=69593

but I get the same issue

clsergent, 2024-06-16:
Well it seems that you switched to KineticNCBeamicon2 post processor (per "_22:29:42 post: KineticNCBeamicon2_") . Could you try again with the Snapmaker postprocessor and report if any error occurs ?

datamystic, 2024-06-16:
Got it - it completes generation without error now. The file doesn't seem to include any actual CNC instructions, I must have done something wrong with the tool setup. I will check some videos and see what I have done wrong! Thanks for your help!

closed 2024-06-16 by datamystic