clsergent/Snapmaker2Postprocessor #16: With FC1.0: Failed to write to output file

ansgar02, 2025-01-26:
Once the version check has been changed to (likely) support FC 1.0 this seems not the only change required but now my understanding of the postprocessing architecture is definitively insufficient:
I get a “Permission denied” and the file that is going to be opened for writing is “-” .
This looks a bit to me like a pipe (with a process at the other end), that should be the output channel, but my environment interprets it obviously as a filename:
Here is the console log:

23:09:41 Running the Python command ‘CAM_Post’ failed:
Traceback (most recent call last):
File “C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Post\Command.py”, line 188, in Activated
post_data = postprocessor.export()
^^^^^^^^^^^^^^^^^^^^^^
File “C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Post\Processor.py”, line 321, in export
gcode = self.script_module.export(sublist, “-”, self._job.PostProcessorArgs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “C:/Users/ansga/AppData/Roaming/FreeCAD/Macro\Snapmaker_2_CNC_post.py”, line 750, in export
return post.export(objects, filename, argstring)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “C:/Users/ansga/AppData/Roaming/FreeCAD/Macro\Snapmaker_2_CNC_post.py”, line 743, in export
with open(filename, ‘w’) as file:
^^^^^^^^^^^^^^^^^^^

[Errno 13] Permission denied: ‘-’

I’m using FC 1.0 on Win11 btw.

Cheers
Oerly

clsergent, 2025-01-27:
Hi, thanks for reporting this bug.

I experienced this error or the first time after downloading v1.0 version. However, it seems to be quite old if I refer to other postprocessors handling it (not all though, **marlin** at least failed).

Anyway,  `export function` follows this scheme:
```
export(objects, filename: str, argstring: str)
```
For any reason, `filename` argument is `'-'` no matter what was entered in the **PostProcessorOutputFile** field. 
I added an extra check to handle this in commit 9b0db397080e81be01407513a419c32c12a793be. 

Please report if it fixed the issue.

ansgar02, 2025-01-27:
Thanks a lot for this immediate turn around - overnight! 
My posting to the freecad forum was answered overnight as well and Larry Woestman pointed out, that the postprocessor API was changed in that sense, that a filename of "-" is now treated as "don't write any output" (cmp. https://forum.freecad.org/viewtopic.php?p=806901#p806901). Not even all postprocessors in the std package have been adapted to this (cmp. Marlin), but for example linuxcnc had it.
If your target is moving it is not a bug....

Cheers
Oerly
P.S. Will test it later during the day

clsergent, 2025-01-27:
I read carefully the forum thread, but I think there is still a bug.

**PostProcessorOutputFile** field is not passed to the `export` function anymore on my computer. '-' is yield no matter what value is provided.

I understand the aim of '-' but it should only appears if no filename is provided. Do you agree with my understanding?

ansgar02, 2025-01-27:
Hi,Yes, I agree in principle.I found it kind of strange, that the general setup for a job asks for an output file, but that name seems not to be passed to the export function of the post processor.But now, the pops up a file selection dialog, asking for the name of the output file.This finally works but doesn’t feel very consistent.Cheers

clsergent, 2025-01-27:
I close the issue as the postprocessor has been fixed. Further inquiries should be addressed to the FreeCAD team.

closed 2025-01-27 by clsergent