clsergent/Snapmaker2Postprocessor #17: Snapmaker doesn't take the NC file...

ansgar02, 2025-01-27:
After using the new postprocessor for a simple little model I got my nc files at the respective location.
I could load them into the Luban workspace and the model and operation are shown in the preview.
When I try to start the carving, a dialog pops up, telling the the jobtype (gcode file type) is inconsistent with the toolhead (3d Printing?? - the aktual toolhead is 50W CNC as shown in the Luban Window...) and that the results may be unwanted (see attachement).

When I try to download the gcode to the machine, both Luban and the Touchscreen confirm successful transmission but the file is not available on the touchscreen (works with a GCode generated by Fusion 360).

Starting the file from Luban results in "noop", i.e. the machine doesn't move at all.
I can't seen any obvious thing in the gcode (a VERY long line containing a bitmap was removed manually to test this for the cause, but that didn't change the behaviour).

I'm attaching the popup, when trying to start the file from Luban and the FC model and the generated gcode in a zip

[FreeCad Projekte.zip](https://github.com/user-attachments/files/18564097/FreeCad.Projekte.zip)

Thanks for any help!

[image]

clsergent, 2025-01-27:
Hi,

Thanks for reporting this issue. 
I will check this and come back to you. 
In the meantime, if you want to explore producted GCODE without the bitmap, you can pass ˋ--no-thumbnail` as argument to the postprocessor.

clsergent, 2025-01-27:
I tried your file, leading me to two conclusions:
- GCODE is ok, it loaded without errors in CAMotics, so it should be safe to use
- Snapmaker changed at some point their GCODE preamble to include various info, such as operation type (CNC, laser, printing) instead of relying on file extension as they used to.

Quick fix (which I did not test as I can't use my machine right now): add this Comment line at the top of your GCODE file:
```
;header_type: cnc
;tool_head: standardCNCToolheadForSM2
;machine: Snapmaker 2.0 A350
;gcode_flavor: marlin
;max_power: 255
```

I just pushed [2164ebc](https://github.com/clsergent/Snapmaker2Postprocessor/commit/2164ebc103a8f295cea1a168e2870768909ad6e7) to do this. 

I will work to adapt other info as they are rather basic (axis maxima, toolhead, etc.). Estimated time will be more tricky

ansgar02, 2025-01-28:
Unforunately the problem seems to be sitting in front of the screen - again....

The output of your post processor is perfect!

I gave it a try and copied section by section from the generated file (that didn't work) into a reduced version of the Fusion generated file that worked and tried to load the result into the machine/TS.
All incremental changes worked fine until both files were equal from a content perspective!
Only difference: The FILENAME SUFFIX!! Changing the ".nc" into ".cnc" turned the file recognized by the TS!!
Currently the machine is carving a little piece of beach-wood!!!

Thanks again for all your help!
From my perspective this issue is solved! 

Cheers

clsergent, 2025-01-28:
Glad to read that everything is alright. 
I wasn't worried by the generated GCODE (checked by CAMotics) but by Snapmaker implementation which has already changed without notice. 

Anyway I will upgrade the header to add metadata to feed Luban, so this issue remains relevant!

I'm now closing this issue.

closed 2025-01-28 by clsergent