clsergent/Snapmaker2Postprocessor #18: Encoding Error?

MxRemy, 2025-03-13:
On FreeCAD 1.0.0, on Windows 10, when I select this post processor I get the following error:

```
14:29:40  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\CAM\Path\Main\Job.py", line 526, in onChanged
    self.tooltipArgs = processor.tooltipArgs
                       ^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/CCL VR - HTC Vive/AppData/Roaming/FreeCAD/Macro\snapmaker_post.py", line 511, in tooltipArgs
    return self.parser.format_help()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 2595, in format_help
    return formatter.format_help()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 287, in format_help
    help = self._root_section.format_help()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 217, in format_help
    item_help = join([func(*args) for func, args in self.items])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 217, in <listcomp>
    item_help = join([func(*args) for func, args in self.items])
                      ^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 217, in format_help
    item_help = join([func(*args) for func, args in self.items])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 217, in <listcomp>
    item_help = join([func(*args) for func, args in self.items])
                      ^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 546, in _format_action
    help_text = self._expand_help(action)
                ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\argparse.py", line 643, in _expand_help
    return self._get_help_string(action) % params
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
<class 'ValueError'>: unsupported format character ')' (0x29) at index 14
```

I looked through for non-ASCII characters in the file name, model tree and constraints, but didn't see any. I could just be missing it somewhere though. For the most part, nothing in particular is named anything other than what FreeCAD auto-names it.

clsergent, 2025-03-13:
Thanks for opening this issue. Your error seems related to how `argparse` format help. There was a percent symbol (%) in the help string of the `--spindle-percent` argument which caused `argparse` to fail (because it relies on this symbol for string interpolation).

I just fixed it, you should not face this issue again. Would you mind reporting if everything is alright now?

MxRemy, 2025-03-13:
Getting closer! The previous error is gone. When I post the process now, I only get this new error, which doesn't prevent me from ending up with a file at least.

```
18:18:26  No toolhead selected, using default (50W CNC module). Consider adding --toolhead
18:18:26  Boundary check skipped: no valid boundaries supplied
```

I didn't see anywhere to configure the toolhead, but this Artisan has a 200W module on it, not sure if it's super important or not. However, the machine is still not recognizing the file, with either .nc or .gcode extensions. It's progress though!

clsergent, 2025-03-14:
Please consider reading the wiki which explains how to add arguments (such as ˋ--toolhead` and ˋ--machine` here). 

If you use the 200W module with Artisan, you should at least add ˋ--machine=artisan --toolhead=200W`

About file extension, the controller filter files depending on the recognized toolhead:
- 3D printing : .gcode
- CNC : .cnc

MxRemy, 2025-03-17:
Ah sorry, I knew you'd mentioned changing the extension, but I forgot which and chose wrong. I'll read the wiki again, must have missed the bit about post processor arguments. Sorry for all the trouble!

So, after fixing my obvious-in-retrospect errors, I am finally running the job on the Artisan. It's a long one and it's in progress now, so I can't give a definitive statement on whether the whole process has any issues yet, but it looks to me as if it's working just fine. I think you can safely say it's tested and working on the Artisan now, other than having to change the extension, which would probably be obvious anyway to someone more experienced than me.

Is there anything in particular you'd like me to test on this machine?