clsergent/Snapmaker2Postprocessor #20: Post Processing Fails to Post (Machine Argument Not Defined)

SaazyTheGoat, 2025-05-26:
- OS: Linux, EndeavourOS
- Desktop Environment: KDE
- FreeCAD Version: AppImage, v1.1.0dev, 2025/05/23
- Architecture: x86_64

Error during post. Argument Required: --machine. The other post processing files work fine. Generic Marlin exports correctly and generates a gcode file with commands.

```
2freecad: error: the following arguments are required: --machine
22:22:43  Running the Python command 'CAM_Post' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAaDlNCK/usr/Mod/CAM/Path/Post/Command.py", line 216, in Activated
    post_data = postprocessor.export()
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 836, in export
    (flag, args) = self.snapmaker_process_arguments()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 482, in snapmaker_process_arguments
    (flag, args) = Path.Post.UtilsArguments.process_shared_arguments(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/Mod/CAM/Path/Post/UtilsArguments.py", line 688, in process_shared_arguments
    args = parser.parse_args(shlex.split(argstring))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/lib/python3.11/argparse.py", line 1874, in parse_args
    args, argv = self.parse_known_args(args, namespace)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/lib/python3.11/argparse.py", line 1907, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAaDlNCK/usr/lib/python3.11/argparse.py", line 2156, in _parse_known_args
    self.error(_('the following arguments are required: %s') %
  File "/tmp/.mount_FreeCAaDlNCK/usr/lib/python3.11/argparse.py", line 2640, in error
    self.exit(2, _('%(prog)s: error: %(message)s\n') % args)
  File "/tmp/.mount_FreeCAaDlNCK/usr/lib/python3.11/argparse.py", line 2627, in exit
    _sys.exit(status)
```

SaazyTheGoat, 2025-05-27:
So, using the --machine argument does make the error go away. It's replaced with a new error to utilize the --toolhead argument. Once both of those are added, I get a PostUtils is not defined error.

`2Spindle speed will be controlled using using percentages.
18:12:38  Running the Python command 'CAM_Post' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAMDFApE/usr/Mod/CAM/Path/Post/Command.py", line 216, in Activated
    post_data = postprocessor.export()
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAMDFApE/usr/Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 838, in export
    return self.snapmaker_process_postables(filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAMDFApE/usr/Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 611, in snapmaker_process_postables
    gcode = self.export_common(objects, filename.format(name=name))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAMDFApE/usr/Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 814, in export_common
    dia = PostUtils.GCodeEditorDialog()
          ^^^^^^^^^

name 'PostUtils' is not defined`

**Revision Number:** 41837 (GIT)

I can try building the freecad-git package from source tonight over the AppImage.

clsergent, 2025-05-28:
Ok this definitely an API change, your revision number is way ahead of mine. 
I will dig further and make the appropriate changes. 

In the meantime, if you need the post processor, you can try the old one which is not built on top of the post processing API, or use a non beta version of FreeCAD.

ipeco85, 2026-09-06:
Hi, I just wanted to use FreeCAD for CNC machining on my Snapmaker. I own both an Artisan and an A250 machine. I use FreeCAD for years, but i'm new to the CAM module, now really disappointed with the limited CNC features of the Luban software. I successfully went through some CAM learning videos, but failed with the snapmaker output - see added snapshot. I use FreeCAD 1.1.0 from 26/03/25 with standard installation on a WIN 11 desktop. Please advise how to overcome the problems.  Installing "machines" as add-on didn't change any issues
Many thanks in advance.
Peter

Type Notifier Message 

* Running the Python conmmand "CAM_Post' failed: 
Traceback (most recent call last): 
File "C:\Program Files\FreeCAD 1.1\Mod\CAM\Path\Post\Command.py", line 217, in Activated 
post_data = postprocessor.export) 

File "C:\Program Files\FreeCAD 1.1\\Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 837, in export 
(flag, args) = self.snapmaker_process_arguments(} 

File "C:\Program Files\FreeCAD 1.1\\Mod/CAM/Path/Post/scripts/snapmaker_post.py", line 483, in 
snapmaker_process_arguments 
(flag, args) = Path.Post.UtilsArguments.process_shared_arguments( 

File "C:\Program Files\FreeCAD 1.1\Mod\CAMPath\Post\UtilsArguments.py", line 773, in process_shared_arguments 
args = parser.parse_args(shlex.split(argstring)) 

File "C:\Program Files\FreeCAD 1.1\bin\Lib\argparse.py", line 1874, in parse_args 
args, argy = self.parse_known_args(args, namespace) 

File "C:\Program Files\FreeCAD 1.1\bin\Lib\argparse.py", line 1907, in parse_known_args 
namespace, args = self._parse_known_argslargs, namespace) 

File "C:\Program Files\FreeCAD 1.1\bin\Lib\argparse.py", line 2156, in _parse_known_args 
self.error_(‘the following arguments are required: %s‘) % 
File "C:\Program Files\FreeCAD 1.1\bin\Lib\argparse.py", line 2640. in error 
self.exit(2, _(%(prog)s: error:%(message}s\n') % args) 
File "C:\Program Files\FreeCAD 1.1\bin\Lib\argparse.py", line 2627, in exit 
_sys.exit(status) 
2 

* freecad.exe: error the following arguments are required: --machine

clsergent, 2026-09-06:
Hi,
- Very naïve question, did you set `--machine=A250` (respectively `Artisan`)? Check https://github.com/clsergent/Snapmaker2Postprocessor/wiki/FAQ#how-to-add-a-post-processor-argument for how to set this argument. 
I definitely need to find some time to check the new postprocessor API...

ipeco85, 2026-09-06:
Hello, thanks a lot for your quick response! Indeed I didn't know that any arguments have to be added. Using the 'machine' and "toolhead" arguments the output seems to come without errors. All other arguments have to be evaluated, not every one sounds familiar to me ..
Again many thanks and have a nice evening!